davidperrenoud/Gamegirl #13: Project still alive ?

darkworks, 2021-01-30:
am wondering if this project is still alive ? or dead , can we have update on v0.2.1 , also can  u guys share gerbers files of v0.2.1 so that we can make pcb 

thanks

32teeth, 2021-01-31:
> Pi 0 is becoming scarce.
Most projects like this have moved to [ESP32](https://github.com/retro-esp32/RetroESP32)

If you are looking for *'nix' based computing, many options now for Pi4 and CM4 coming around the corner

Depending on your spend, ESP32 offers instant boot

davidperrenoud, 2021-02-05:
I have not worked on this project for a long time, but the Lakka build that was done for the Gamegirl is more or less similar to what is now available for the GPi Case:
http://retroflag.com/GPi-CASE.html
http://lakka.tv/get/linux/rpi/

Alternatively, there are some pretty nice consoles from Hardkernel/ODROID nowadays.